ARMmbed/pal #15: remove use of tabs and be consistent with use of white spaces

rgrover, 2016-10-14:
The use of tabs results in poor readability in general except when the user's editor-settings match that of the developer. Please remove the use of tabs. One example is `pal_errors.h` where the enumeration for `palError_t` looks odd with nearly every tab setting. Replace tabs with spaces.

Please also be consistent with the use of white spaces. Here's what I see within `palError_t`. The use of white spaces around `=` is inconsistent.

```C++
	PAL_ERR_SOCKET_DNS_ERROR = 								PAL_ERR_SOCKET_ERROR_BASE + 16,			/*! DNS lookup error*/
	PAL_ERR_SOCKET_HDCP_ERROR = 							PAL_ERR_SOCKET_ERROR_BASE + 17,			/*! HDCP error*/
	PAL_ERR_SOCKET_AUTH_ERROR = 							PAL_ERR_SOCKET_ERROR_BASE + 18,			/*! authentication error*/
	PAL_ERR_SOCKET_OPTION_NOT_SUPPORTED = 					PAL_ERR_SOCKET_ERROR_BASE + 19,			/*! socket option not supported*/
	//update Error
	PAL_ERR_UPDATE_ERROR_BASE 			= 							(-1 << PAL_ERR_MODULE_UPDATE),			/*! generic error */
	PAL_ERR_UPDATE_ERROR      			= 							PAL_ERR_UPDATE_ERROR_BASE,  			/*! unknown error */
	PAL_ERR_UPDATE_BUSY   	   			= 						    PAL_ERR_UPDATE_ERROR_BASE + 1, 			/*! unknown error */
	PAL_ERR_UPDATE_TIMEOUT       		= 							PAL_ERR_UPDATE_ERROR_BASE + 2,		    /*! unknown error */
```

NirSonnenschein, 2016-10-18:
Agreed, we will clean this up for the upcoming release

NirSonnenschein, 2016-10-26:
Fixes in R.10

closed 2016-10-26 by NirSonnenschein