ARMmbed/pal #17: documentation around pal_osKernelSysTick is unclear

rgrover, 2016-10-14:
Reading the [doc-header for pal_osKernelSysTick](https://github.com/ARMmbed/pal/blob/master/Source/PAL-Impl/Services-API/pal_rtos.h#L77), I could not figure out what the API was meant to return.

Tick could be an overloaded concept here. Assuming that a 'tick' refers to the interval of time between the firing of SysTick hardware interrupts, as in ARM cores, I struggled to understand whether pal_osKernelSysTick() requires returning the instantaneous value of the SysTick h/w counter or the count of 'ticks seen so far' as maintained by the kernel.

On many CortexM micros, the Systick is a 24-bit down counter running out of the core CPU clock. Is this API meant to return the value of this SysTick timer? If so, the following comment in the documentation made it confusing for me:

```
 \note this counter will wrap around very often (e.g. once every 42 sec for 100Mhz)
```

The thing about wrapping once every 42 sec at 100Mhz is true only for 32-bit hardware counters. But SysTIck is not a 32-bit counter.

Perhaps I'm wrong, but I believe there is at least a source of confusion here which should be clarified.

rgrover, 2016-10-14:
Please define some bounds on what a `tick` could be. Without this, a platform porter could choose to define something as precise as a clock cycle as a `tick`, alternatively the platform porter could choose `tick` to be a very imprecise interval.

Please also clarify what this API is meant to be used for.

mohammad1603, 2016-11-06:
Thanks for the comment, we added more information / specifications about system tick definition and its bounds in the PAL RTOS documentation.
in general this API meant to return the kernel system tick counter, regarding the implementation of the tick counter (whether h/w or s/w) it depends on the platform implementor, our requirment is to limit the lower bound to 1KHz at least.
The usage of this API is for time comparison in general.
the clarifications will be in the upcoming PAL release.

closed 2018-04-18 by teetak01