ARMmbed/pal #18: tests assume that pal_osKernelSysTick() returns non-zero

rgrover, 2016-10-14:
refer to https://github.com/ARMmbed/pal/blob/master/Test/Unitest/pal_rtos_test.c#L57

It should be acceptable for pal_osKernelSysTick() to return 0.

rgrover, 2016-10-14:
this is particularly true since the definition of a `tick` is not clear. If the platform implementor defines a `tick` to be an interval of around 10ms (which is a common value), then pal_osKernelSysTick() will return the value 0 for the first 10ms of the lifetime of the test program, and then this [above-mentioned](https://github.com/ARMmbed/pal/blob/master/Test/Unitest/pal_rtos_test.c#L57) assertion will fail.

NirSonnenschein, 2016-10-26:
fixes for the tests are deployed n R1.0
Some additional documentation regarding the tick count was added, more may be needed

closed 2016-10-26 by NirSonnenschein