ARMmbed/pal #19: documentation for pal_osTimerCreate() does not explicitly specify allocation semantics

rgrover, 2016-10-17:
When memory isn't available, an implementation has the choice of either waiting (possibly indefinitely) for memory or returning `PAL_ERR_NO_MEMORY`. It is not clear from the documentation if (and under what circumstances) pal_osTimerCreate() is allowed to block waiting to allocate a kernel timer.

I also believe that the documentation should make it explicit that `PAL_ERR_NO_MEMORY` shall be returned in cases where memory resources have been exhausted and the system chooses not to wait.

NirSonnenschein, 2016-10-18:
I agree ( I dont think we ever considered a port taking the waiting approach), we will add this to the documentation.
I can't really think of a reasolnable port that blocks wating for resources (unless the underlying OS API is blocking [which would be strange in itself] and they have no choice). this is probably why we didn't think to document this

mjs-arm, 2016-10-19:
Waiting for memory should not be permitted, the porting layer should be required to return no_memory. The other approach makes it very difficult to prove dead lock is not possible

NirSonnenschein, 2016-10-19:
I think all of us are agreed on this point, we will doncument this better.

NirSonnenschein, 2016-10-26:
added some comments and integration guide notes regarding the behavior APIs and waiting for resources (Implementaiton should not wait for resources).

closed 2016-10-26 by NirSonnenschein