ARMmbed/pal #20: pal_osTimerCreate() does not specify the context for the invocation of function callback

rgrover, 2016-10-17:
From the documentation of pal_osTimerCreate(), it is not clear to the platform porter or the user under what context the function callback will be invoked when a timer expires. Will the callback be called from a task? or an interrupt?

NirSonnenschein, 2016-10-18:
this very much depends on the underlying implementation (which will proabably use underlying os funcitonality). In some cases I have encoutered the OS API for timers doesn't give you any controll of the context from which the callback is called (e.g. CMSIS specifies timers execute in the osTimerThread but POSIX documentation doesn't clearly specify which context the timer callback is run from [unless I missed sonething] ). The programmer writing the service should not assume that the callback is callsed from a particular context because the PAL doesn't guaranteed this and this is up to the porting layer//underlying OS to decide.
I agree that this could possibly cause issues in certain cases, but I'm not sure if we really have a choice.

rgrover, 2016-10-19:
If behaviour varies with the platform, it might help to make a statement about the most constrained implementation. Perhaps the user can be informed that they should not use callbacks which need large computing or memory resources. Perhaps the documentation can state that the platform is allowed to implement this using interrupt-handlers.

NirSonnenschein, 2016-10-19:
Hi Rohit,
I agree regarding documenting the minimal scope of callbacks.
Normally there is some OS functionality that allows this implementation so there shouldn't be a need to use interrupt handlers. I wouldn't want to encourage this type of implementation if there is an alternative as it can cause all sorts of unexpected problems (some APIs [even short ones] may not work in the context of an interrupt handler).

Thanks,
                Nir S.

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

mjs-arm, 2016-10-19:
Rohit makes good points here. The contract between the application and the porting layer through the PAL API needs  much tighter specification.   Encourage /discourage is not good enough.  The application programmer needs to know what primitives they are permitted to use on the callback, the platform layer needs to know what primitives they must be able to honour on the callback thread they provided whether it is interrupt , finer, task whatever.  There are other issues with this interface beyond the context of the callback.  The callback thread is owned by the porting layer, but the work executed on that thread is owned by the application, there is no mechanism for them to coordinate stack size.

rgrover, 2016-10-19:
In addition to the stack-size, the scheduling priority of the `callback thread` is also open to arbitrary selection.

closed 2018-04-18 by teetak01