ARMmbed/pal #21: Not clear if pal_plat_osTimerStart() should accept the value 0 for the `millisec` parameter

rgrover, 2016-10-17:
What should the system do if a timer is started with a timeout value of 0? Is this a param error? or should the function callback be invoked immediately?

rgrover, 2016-10-17:
In more general terms, depending upon the definition of a `tick`, the documentation should clarify the platform's behavior when the `millisec` parameter specifies a timeout which is smaller than (or equal to) a single tick.

mohammad1603, 2016-11-06:
Thanks for the comment,  we added more info to the documentation for the cases:
1- single system tick interval.
2- in timeout value 0 case, the function returns "PAL_ERR_RTOS_VALUE" error. 
documentation fix will be available in the upcoming PAL release.

closed 2018-04-18 by teetak01