ARMmbed/pal #22: pal_osTimerDelete() has the side-effect of overwriting its parameter

rgrover, 2016-10-17:
```C++
/*! Delete the timer object
*
* @param[inout] timerID: the handle for the timer to delete, in success:(*timerID = NULL).
*
* \return PAL_SUCCESS when timer deleted successfully.
*		  PAL_ERR_RTOS_PARAMETER when timerID is incorrect.
*/
```

pal_osTimerDelete() takes its timerID as an in/out parameter, and requires the implementation to overwrite the timerID with NULL upon success. This is a bit surprising. At the very minimum, it means that the caller's variable holding the timerID has to be mutable---i.e. it cannot be placed in a `const` owning structure. More importantly, this is an un-necessary side-effect and does not have precedent. Users of free() don't expect similar semantics.

This is not a serious issue, but I request that the designers should rethink the approach. Modifying caller supplied memory may suffer from strange consequences such as crossing access/protection privileges.

rgrover, 2016-10-19:
The same is true for `pal_osPoolDestroy()`.

mohammad1603, 2016-11-07:
Thanks for the comment, the sequnece for these APIs and similar APIs is as follow:
* `pal_os$(OBJECT)Create(..., Pointer to hold Object's handle);` this pointer will contain PAL allocated memory of the requested object.
* `pal_os$(OBJECT)Start/Wait/Release(Object's handle);`
* `pal_os$(OBJECT)Delete(Pointer holds Object's handle); `-> PAL will free the PAL allocated memory in the Create API and assign NULL to this pointer. 
in this way we prevent from accessing invalid memory address in case application try to use the handle again after deleting the `$(OBJECT)`.

PAL requests a mutable variable (for the create function) and this function deletes the object from the underlying platform. so users should not expect same behavior as `free()` in this case.

rgrover, 2016-11-07:
ok. Thanks for the clarification. If this is the general design being applied, then it is acceptable.

closed 2016-11-07 by rgrover