ARMmbed/pal #23: there should be a way to allocate osTimers statically

rgrover, 2016-10-17:
the pal_osTimerCreate() and pal_osTimerDelete() are designed for implementations based on memory-allocation. There should also be a way to allocate osTimers using static memory.

Dynamic allocation of timers is useful, but then this should also be accompanied by a clear description of  associated memory allocation model, and these model/policies should be under the control of the user. 
Static allocation of timers will result in a smaller impact on resources. 

Additionally, if static memory is used then APIs such as pal_osTimerDelete() won't assume being able to overwrite the timerID parameter supplied by the caller because callers could be using `const static` containers to hold timerIDs [refer to https://github.com/ARMmbed/pal/issues/22].

NirSonnenschein, 2016-10-19:
originally all RTOS structures were statially allocated only (this is the original Architectural design by Erez L.) however due to demands by the services we were asked to change all of the APIs to dynamically allocating versions (this makes some sense but also has some downsides obviously). in the specific case of timers, Niehter the CMSIS nor the POSIX timer APIs allow you to statically allocate stack for a timer.
As this is the way timerd work on most systems, so I'm not sure we can change this without making the porting efforts for PAL much more difficult.

rgrover, 2016-10-28:
The argument for static allocation is also true for message-queues, tasks, mutexes, etc. These could also benefit from static definition/allocation.

closed 2018-04-18 by teetak01