ARMmbed/pal #25: issuing callbacks based on timer-expiry is potentially risky

rgrover, 2016-10-18:
`pal_osTimerCreate(callbackFunc, ...)` and `pal_osTimerStart()` allow applications to schedule future invocations of arbitrary callback functions. The invocation of these callbacks then becomes the responsibility of the platform. This is potentially risky because callbacks can be arbitrarily complex in their use of resources (stack and processing-time).

As an example of this problem, if the platform implementor chooses to deploy a dedicated callback-thread to process these callbacks, (s)he would need to make an arbitrary decision about the stack-size of this callback-thread (and its priority). No particular choice of stack-size or priority is guaranteed to work in the general case. This is potentially insecure.

rgrover, 2016-10-18:
related to https://github.com/ARMmbed/pal/issues/20

NirSonnenschein, 2016-10-19:
Hi Rohit, 
I agree this can be a problem, however this is the way timers are implemented on most systems I have enountered so far. I'm really not sure we can do anything about this at the PAL level without making the porting efforts much more difficult. 
My personal approach to timers is that they should be used very rarely and only for cases when absolutely required. 
the Mbed services do use timers successfully however (e.g. connector ) so YMMV.

NirSonnenschein, 2016-10-26:
not sure this can be fixed at the PAL level

closed 2016-10-26 by NirSonnenschein