ARMmbed/pal #26: pal_osTimerDelete() does not specify how the platform should handle active timers

rgrover, 2016-10-18:
The documentation for `pal_osTimerDelete()` does not clarify the platform's behaviour when called for timers which are active.

There are two reasonable options:
- The active timer would automatically be stopped before getting deleted.
- An error should be returned to the user to indicate potential resource leak.

NirSonnenschein, 2016-10-19:
I agree that timer delete should stop timers before deletion. I think this is the best option.
We will fix this for the next version.

NirSonnenschein, 2016-10-26:
Mohammad checked this out, and apperantly the CMSIS API documentation guaratess that the delete funciton will stop the timer before deletion. this means that the MBED OS port should behave ok without modifications. 
We will add a delete in all ports where the delete doesn't guaranteed this behavior

rgrover, 2016-10-26:
I don't think you should unilaterally close a bug report. I am not satisfied that the right thing has been done in this case. The documentation doesn't specify platform's behaviour when delete is called on an active timer. MbedOS might be doing the right thing, but unless the expected behaviour is documented other ports might not even know how to handle this situation.

NirSonnenschein, 2016-10-27:
Agreed regarding documentaiton, re-opning the issue so that we dont' forget.

mohammad1603, 2016-11-07:
We added this comment as a requirement from the platform layer, this API must stop the timer before deletion.
Thanks for the comment, the updated API will be available in the upcoming PAL release.

closed 2018-04-18 by teetak01