ARMmbed/pal #31: errors in the API documentation for pal_osMessageGet()

rgrover, 2016-10-21:
```C++
/*! Get a Message or Wait for a Message from a Queue.
*
* @param[in] messageQID the handle for the memory pool
* @param[in] timeout timeout in milliseconds
* @param[out] event the data to send
*
* \return the function returns the status in the form of palStatus_t which will be PAL_SUCCESS(0) in case of success and one of the following error codes in case of failure:
* PAL_ERR_RTOS_RESOURCE - in case semaphore was not available but not due to time out.
* PAL_ERR_RTOS_TIMEOUT -  no message arrived during the timeout period.
* PAL_ERR_RTOS_RESOURCE -  no message received and there was no timeout
*/
palStatus_t pal_osMessageGet(palMessageQID_t messageQID, uint32_t timeout, uint32_t* messageValue);
```

- The third parameter in the API is `messageValue`. This is inconsistent with the documentation.
- In the part describing return values, there is duplicate mention of `PAL_ERR_RTOS_RESOURCE`.
- In the first instance of PAL_ERR_RTOS_RESOURCE, the document mentions `in case semaphore was not available but not due to time out.` Why is there a mention of `semaphore`? How is that related?
- What is the meaning of `no message received and there was no timeout`? When can this happen?  And how is it different from `no message arrived during the timeout period`?
- The documentation does not specify behaviour when timeout is provided as `0`.

rgrover, 2016-10-27:
Can `pal_osMessageGet` be called from an interrupt context if timeout is 0?

rgrover, 2016-10-27:
The documentation also does not specify if `PAL_RTOS_WAIT_FOREVER` is a valid argument for `timeout`.

mohammad1603, 2016-11-01:
Thanks for the comment, the fix will be published in the upcoming PAL release

closed 2018-04-18 by teetak01