ARMmbed/pal #33: pal_osThreadCreate() supports multiple thread-specific arguments

rgrover, 2016-10-26:
pal_osThreadCreate() allows for both `void* funcArgument` and `palThreadLocalStore_t* store`. This seems redundant. Why isn't any one of the above two sufficient?

mohammad1603, 2016-11-07:
The `funcArgument` is meant to be used as an argument for the thread function.
regarding the TLS (Thread Local Store) it is meant to be used for many ways of configurations/ communectations / data-sharing between different threads. For example, defining globals between threads.

rgrover, 2016-11-07:
Why isn't any one of the above two sufficient? Why do we necessarily need both?

Zephyr is an example of a platform where only TLS is available, and the function-argument needs to be simulated.

mohammad1603, 2016-11-07:
There are different approaches to API design. In some cases it is more clear to separate parameters
by emphasizing their functionality difference. In OS's like Zephyr the irrelevant parameter can simply accept NULL.

rgrover, 2016-11-08:
In Zephyr, there is *no* support for the function argument. The entry function for a task *does not* receive a parameter. This needs to be simulated when porting PAL to that platform. It may only be a design choice  for PAL to include the function argument, but it turns into a porting hassle for a platform porter. Please rethink the design if possible.

mohammad1603, 2016-11-08:
The current design is giving a wide flexibility to the application code and to the porting side. 
according to your comment, the application code will need to work _twice_:
1- packing the TLS and the thread function argument (probably they will need to add new data type for this).
2- unpacking this parameters again inside the thread function code, assuming that the porting done as expected (because combining these two probably will lead many issues), not mentioning the complexity of these design will be added to the code.

Therefore the current design is giving more clarity and flexibility to the application side and the porting as well.
Thanks again for the comment and I think this issue can be closed.

rgrover, 2016-11-08:
ok. If you think this is better overall, then we can close this issue.

closed 2016-11-08 by rgrover