ARMmbed/pal #41: checks in pal_socket_test.c incorrectly use reversed arguments

rgrover, 2016-11-01:
`pal_socket_test.c` repeatedly uses the following pattern:

```C++
result = someAPI();
TEST_ASSERT_EQUAL(result, PAL_SUCCESS);
```

The Unity macro TEST_ASSERT_EQUAL is defined as `TEST_ASSERT_EQUAL(expected, actual)`. The arguments are the reverse of how it is used in pal_socket_test.c.

This produces the following confusing output in the case of a failure:

```
TEST(pal_socket, basicTCPclinetSendRecieve).../update_client/src/pal_socket_test.c:164:TEST(pal_socket, basicTCPclinetSendRecieve):FAIL: Expected -65522 Was 0
unity_assert_failure
```

The correct output would have said: `Expected 0 Was -65522`

mohammad1603, 2016-11-01:
Thanks for the comment, we are looking at this issue and it will be fixed in the upcoming PAL release.

rgrover, 2016-11-01:
@mohammad1603 @NirSonnenschein : thanks for acknowledging these issues.

It would be very helpful for me if you make small commits to address these issues individually.

As a submitter of these bug reports and also as a platform porter, I would like to understand and synchronize with your changes to PAL. If you update an API (for instance), it would be highly desirable if I am able to see a small diff showing what has changed. Small, separate diffs will also alert me If the changes require me to update the platform-port I'm working on.

If, on the other hand, you submit one massive new release combining fixes to several issues (including changes to auto generated doxygen), it will be very difficult for me (the bug reporter) to isolate changes into separate topics and make sense of everything.

The standard way to accomplish this is to make small commits to the PAL repository and associate them with issue-IDs. [Github allows for this](https://help.github.com/articles/closing-issues-via-commit-messages/).

avolinski, 2016-11-02:
Hi Rohit,

I totally understand the benefits of gradual small commits per issue.
The thing is that currently PAL releases in public repository are issued only in milestone versions.
And this means cumulative commits of all features and issues that are planned each time for the upcoming release.
This policy is currently not expected to change. Thus we can only release the changes in bulks.
Of course the release notes will include all the changes.
And the resolved issues will be marked as done right after the release is out.

rgrover, 2016-11-02:
@avolinski: thanks for the clarification

avolinski, 2016-11-02:
In addition if you would like to track the progress per issue.
You are invited to view the private pal master repository

avolinski, 2016-11-02:
And Thank You Very Much for your reviews, we really appreciate it!

closed 2018-04-18 by teetak01