ARMmbed/pal #58: PAL should not call `set_timeout` on mbed OS socket

janjongboom, 2017-04-04:
We're calling `set_timeout` on mbed OS socket in PAL, see [here](https://github.com/ARMmbed/pal/blob/4e46c0e/Source/Port/Reference-Impl/mbedOS/Networking/pal_plat_network.cpp#L352).

I believe this does not what PAL expects. If you call `set_timeout` then all blocking calls to the socket like `recv()` will fail with -3001 (`NSAPI_ERROR_WOULD_BLOCK`) error code. I've seen this behavior in the Ameba RTL8195A port. Downloading firmware image would stop right after the first block with a `-3001` error code, but commenting out the call ([see here](https://github.com/ARMmbed/TonyEdimaxAirbox/pull/5/commits/aadae9e2b8b4efac5b3518f7df2c9fb0d700d466?w=1)) will make the process continue.

I'm not sure if this is an implementation bug in mbed OS, or some unclear docs, but I've been bitten by it when writing mbed-http as well (that's why I realized what the issue was straight away). Setting set_timeout to any value will make blocking APIs unavailable.

avolinski, 2017-04-05:
Hi Jan,
are you relating to mbedOS5.2 behavior or to later versions?
the most updated code lies in https://github.com/ARMmbed/mbed-client-pal private repo

janjongboom, 2017-04-05:
@avolinski This call is still there in latest mbed-client-pal: [see here](https://github.com/ARMmbed/mbed-client-pal/blob/master/Source/Port/Reference-Impl/OS_Specific/mbedOS/Networking/pal_plat_network.cpp#L716).

The code I linked above is based on a quite recent version.

NirSonnenschein, 2017-04-06:
Hi Jan, 
we call set_timeout as a way to set the timeout for socket operations on a given socket in mbedOS. 
Our tests using MbedOs over K64F use this behavior fro UDP and it seem to perform ok (e.g. in one test we set a timeout and then preform UDP blocking sendTo and recieveFrom , which work ok, and then one recieveFrom  which issupposed to block which blocks). I will add a TCP check to be sure.
maybe the K64f behavior is different for the ports?
do you mean that the timeout should be set suing the setsockopt funciton? does this operate as you would expect on the port?

janjongboom, 2017-04-06:
@NirSonnenschein I've seen this on TCPSocket, not UDP. Would be interested to see what you find out. I'm not sure how the timeout should be set properly...

NirSonnenschein, 2017-04-06:
thanks, Im working on adding a test for TCP blocking. will update as soo as I've run it.
can you provide any more details on the failing scenario? basically all blocking socket operations fail after this call?

janjongboom, 2017-04-06:
@NirSonnenschein Yep, all return -3001 socket would block error.

NirSonnenschein, 2017-04-06:
I just a timeout of 1 second to my basic TCP test (which then performs connect , send and recieve of a little data and the blocking worked as expected). this seems to work fine on MbedOS on the K64F, so I'm guessing that we are dealing with a porting issue.

here is the test scenario with the added timeout:

TEST(pal_socket, basicTCPclientSendRecieve)
{
    palStatus_t result = PAL_SUCCESS;
    palSocketAddress_t address = { 0 };
    const char message[] = "GET / HTTP/1.0\r\n\r\n";
    size_t sent = 0;
    char buffer[100] = { 0 };
    size_t read = 0;
    palSocketLength_t addrlen = 0;
    int timeout = 1000;

    result = pal_socket(PAL_AF_INET, PAL_SOCK_STREAM, false, 0, &g_testSockets[0]);
    TEST_ASSERT_EQUAL_HEX( PAL_SUCCESS, result);

    result = pal_getAddressInfo(PAL_NET_TEST_SERVER_NAME, &address, &addrlen);
    TEST_ASSERT_EQUAL_HEX( PAL_SUCCESS, result);

    TEST_PRINTF("addr: %u %u %u %u \r\n",
        (unsigned char)address.addressData[2],
        (unsigned char)address.addressData[3],
        (unsigned char)address.addressData[4],
        (unsigned char)address.addressData[5]);

    result = pal_setSockAddrPort(&address, PAL_NET_TEST_SERVER_HTTP_PORT);
    TEST_ASSERT_EQUAL_HEX( PAL_SUCCESS, result);
    
    result = pal_setSocketOptions(g_testSockets[0], PAL_SO_SNDTIMEO, &timeout, sizeof(timeout));
    TEST_ASSERT_EQUAL_HEX(PAL_SUCCESS, result);
    result = pal_setSocketOptions(g_testSockets[0], PAL_SO_RCVTIMEO, &timeout, sizeof(timeout));
    TEST_ASSERT_EQUAL_HEX(PAL_SUCCESS, result);

    result = pal_connect(g_testSockets[0], &address, 16);
    TEST_ASSERT_EQUAL_HEX( PAL_SUCCESS, result);
    result = pal_send(g_testSockets[0], message, sizeof(message) - 1, &sent);
    TEST_ASSERT_EQUAL_HEX( PAL_SUCCESS, result);
    result = pal_recv(g_testSockets[0], buffer, 99, &read);
    TEST_ASSERT_EQUAL_HEX( PAL_SUCCESS, result);

    TEST_ASSERT(read >= 4);
    TEST_ASSERT(buffer[0] == 'H' && buffer[1] == 'T'&& buffer[2] == 'T' && buffer[3] == 'P');
    pal_close(&g_testSockets[0]);

}

closed 2018-04-18 by teetak01