ARMmbed/pal #67: PAL TCP socket fails to connect when using mesh stack

artokin, 2018-01-16:
PAL TCP socket does not connect to connector when mesh stack is used.

PAL is trying to connect a TCP socket to connector and it gets return value NSAPI_ERROR_IN_PROGRESS that will be converted to PAL_ERR_SOCKET_GENERIC in PAL. The error code is returned to application that will close the socket.

Steps to reproduce:
-Use mbed-os-example-client (99d61e4) in Thread or 6LoWPAN mode according to instructions in application

Expected result:
-Application is connected to connector using TCP socket

Actual result:
-Application can't connect. Following log is produced:
```
[EasyConnect] Connected to Network successfully
[EasyConnect] MAC address fc:c2:3d:00:00:04:9b:95
[EasyConnect] IP address 2001:470:6af8:3000:a46f:1980:a5c4:ffb3

SOCKET_MODE : TCP
Connecting to coaps://[2607:f0d0:2601:52::20]:5684

[ERR ][mClt]: pal_connect(): failed: -65536

```

avolinski, 2018-01-16:
Hi @artokin 
the code in this repository serves a previous version of mbed client. which is near a year old.
for the most updated code please refer to restricted repository:
https://github.com/ARMmbed/mbed-client-pal
cc @NirSonnenschein

NirSonnenschein, 2018-01-16:
Hi @artokin , 
this error should be translated correctly on the new version of PAL. please let me know if you still run into any problems.

artokin, 2018-01-19:
OK, thanks. 
I'll try with the new version.

closed 2018-01-19 by artokin